larus-breeze/sw_sensor #61: Accelerometer Calibration

realtimepeople, 2023-04-20:
In has been observed that the accelerometers of the Mti1 are not precisely calibrated.
Offset and gain vary by some 1-2%. At least for the channel(s) that are used
for the aircraft's down axis a calibration will give some improvement for the
AHRS and the variometer.

The calibration can probably only be done manually on the ground.

We shall add a possibility to calibrate the accelerometers similar to the magnetic calibration.

Simsys, 2025-02-01:
This should be done during production and not be expected of the user

MaxBaex, 2025-04-19:
After a discussion with @realtimepeople .  The used Mti1-IMU Accelerometers factory calibration is very good. There will be no further calibration implemented nor usefull.

closed 2025-04-19 by MaxBaex