larus-breeze/sw_sensor #75: Firmware update: Check SW und HW version

MaxBaex, 2024-01-17:
in the update file and compare to running software version before flashing. As described here: https://github.com/larus-breeze/doc_larus/blob/master/documentation/can_details/definitions.md#manufacturer-and-hardware-versions

MaxBaex, 2024-03-05:
[        self.elf_file = ELFFile(in_stream, sys.stdout)](https://github.com/larus-breeze/sw_sensor/blob/a608181e8886fecc0391fd89f32950ce737c1470/sw_stm32/Image_Loader/pack.py#L35)  

second parameter sys.stdout needs to be removed as script fails with
 TypeError: __init__() takes 2 positional arguments but 3 were given

MaxBaex, 2024-03-06:
Use http://elm-chan.org/fsw/ff/doc/findfirst.html     to find all  *.bin images.     Then loop through all *.bin images and
- Match the hardware version in each image. 
- Search for the *.bin image with the "newest/highest software version"   File name doesn't matter except for the file-ending. 
- Compare software version from newest *.bin file to the currently running version.  Flash if the uSD image contains a different version. 

- No role-back protection and no Signature (authenticity) checking right know.

MaxBaex, 2024-08-20:
Where is the hardware and software version of the currently flashed image available?

realtimepeople, 2024-08-20:
The present minimalistic solution already enforces not to flash the same image again.
The GIT commit hash is placed at the beginning of the flash image, immediately behind the interrupt vectors. If this pattern is already in FLASH, the updater refuses flashing.

Simsys, 2025-02-01:
> Use http://elm-chan.org/fsw/ff/doc/findfirst.html to find all *.bin images. Then loop through all *.bin images and
>
> * Match the hardware version in each image.
> * Search for the *.bin image with the [...]

From a system point of view, it makes sense to carry out these tests as described. You can then copy the binaries of all Larus components to an SdCard and update all components. The Larus devices select the image they need.

At the moment there is no protection against flashing the wrong image and then breaking the sensor box. It is really trivial to implement this protection.

The concept for this is described in detail here https://github.com/larus-breeze/sw_tools/tree/master/sw_update

MaxBaex, 2025-04-18:
After a discussion with @realtimepeople a downgrade shall also be possible so the flash process looks for the highest software version on the sd-card and flashs this version if it is diffent to the currently running version.

MaxBaex, 2025-04-18:
The current implementation: https://github.com/larus-breeze/sw_sensor/blob/master/sw_stm32/Communication/uSD_handler.cpp#L480   seems to write a new sd-card image to the flash address 
uint32_t flash_address = 0x80000;   This seems to work but which address is this?     Flash starts at 0x8000000

MaxBaex, 2025-04-19:
understood.  0x80000   is equal to 0x8080000   as 0x0 is mapped to 0x8000000.  The EEPROM issue can be handled as an isolated issue https://github.com/larus-breeze/sw_sensor/issues/117

realtimepeople, 2025-04-24:
I have no idea what all this stuff shall be good for.
The existing way to flash any image called larus_sensor_V2.img if it is not already running fits all needs. It supports upgading and downgrading the firmware and prevents flashing  alien code. What is the use-case for version information in the file name ?

MaxBaex, 2025-04-25:
It implements the ‘software update of Larus components’ concept as described: https://github.com/larus-breeze/sw_tools/tree/master/sw_update We are not developing stand-alone devices, but a system. Each Larus component should behave similarly when it comes to basic functionality such as firmware updates. I would like to have version information in the file name to optimise comprehensibility and complexity for end users.

MaxBaex, 2025-05-20:
merged with #137

closed 2025-05-20 by MaxBaex