larus-breeze/sw_sensor #86: Wind direction is flipping around when speed = 0

Simsys, 2024-03-07:
If you display it like this, it can irritate the user. The sensor box reports sensor values that it cannot determine. In this respect, these values are incorrect. If the aircraft is stationary, no wind calculation is possible. 

You could solve the issue as follows:
1. set the angle of the current wind and the average wind to 0 if no reliable wind calculation is possible
2. define a validity indicator for the wind calculation, which is determined and transmitted by the sensor box. The display devices can then decide what they want to display.
3. you can combine both

realtimepeople, 2024-03-28:
To avoid singularities it may be better to report the wind vectors as north and east components. This can be changed in combination with the migration to float format for the CAN packets. For NMEA a change of the Larus proprietary format would be necessary. This changes would also fix the wind-direction flipping effect.

Simsys, 2024-05-18:
Will be handled in frontend

closed 2024-05-18 by Simsys