larus-breeze/sw_sensor #105: Filename of images should contain their SW version

degeflija, 2024-07-19:


MaxBaex, 2024-08-20:
#75 is a prerequisite as there is currently a hardcoded filename required for all sensor firmware updates.

Simsys, 2025-02-01:
The topic should be discussed further in #75 and this entry should be closed.

MaxBaex, 2025-04-19:
enhance scripts/create-git-info-header.py  so that the file name contains the version information.  But this can only be rolled out once the firmware supports firmware update filenames different to larus_sensor_V2_image.bin

MaxBaex, 2025-04-19:
rename  create-git-info-header.py   as it also creates metadata for firmware update images to
update-git-metadata.py

MaxBaex, 2025-05-17:
Work on this once the issue #117 has been released in a public version.   

- Update  create-git-info-header.py so that it adds the git tag version also in the Image_Loader/pack.toml file to the filename
[image]
name = "larus_sensor_V2_image.bin"

- Update the memory addresses in sw_stm32/scripts/template_pack.toml  according to the new flash layout in #117
- Use the new copy function https://github.com/larus-breeze/sw_tools/tree/master/sw_update/stm32f407_1m-806
- Pack the fimware binary according to the 1.0.0 eeprom layout
- Test if firmware updates are working.

MaxBaex, 2025-05-17:
Update Lib version before creating a pull request.

PR #148 (2025-06-02): Firmware filenames with version and memory layout update
fix  #105 
Documentation how to use the update firmware image creation:  sw_stm32/scripts/README.md