larus-breeze/sw_sensor #106: Inhale and propagate Flarm data stream to XCSoar-type devices

degeflija, 2024-07-23:
Receive Flarm NMEA datastream on RS232 jack and merge these packets into outgoing NMEA datastream created by the LARUS sensor.
This would facilitate showing Flarm targets and Flarm warnings on connected XCSoar type devices without any additional soft/hard ware.
This feature would substitute "IOIOBox"-es.

realtimepeople, 2024-07-23:
To integrate this feature has been planned for the system, but not for the sensor firmware,
Presently we have the idea to prepare a RS232->CAN gateway for this purpose.
Nevertheless it may be an option to support reception of FLARM's NMEA stream on the RS232
and forward it into the CAN data stream.

degeflija, 2024-07-23:
Ad 1 : As far as I know, there is no device on the horizon which might digest Flarm type data on the CAN data stream. So, I do not understand the purpose of a planned R232 - CAN gateway in inbound direction. Outbound, yes, if you have an elaborated frontend device which serves as source for task definition data to be used to set a task in Flarm. But that is also a far fetched idea. Is the purpose of the gateway described somewhere ?

Ad 2 : In addition to my original issue : I was wrong in assuming that Stefly-s machine could benefit from my proposal. SteFly needs the RS232 as Sensor output of NMEA data, as it is not connected via BlueTooth. My proposal is good only for BlueTooth coupled XCSoar devices.
Since these two use cases (SteFly OpenSoar or handheld smartphone XCSoar/OpenSoar) are mutually exclusive, the issue is still worth a thought. It could be implemented via compile time switch or a startup switch ala sensor.readings.