larus-breeze/sw_sensor #107: Competion Mode:  switch off AHRS output

MaxBaex, 2024-10-18:
Full AHRS output is not allowed during competitions. 
The Sensor should therefor provide a feature to switch off the Pitch and Roll output from the AHRS Data for X-Days. It shall not be possible to enable AHRS before the X-Days elapsed. 

- The log file on the SD card shall contain one of the following strings (Timestamps in ISO 8601 Format):  
  - AHRS output disabled until 2024-10-28T08:00Z  
  - AHRS output active

- The X-Days shall be settable via 
  - CAN-Message.  
  - Configuration file on the uSD Card 
- The sensor shall process such a message with the current GNSS time and calculate the date and time until the AHRS output is switched off. This information shall be stored in the STM32s internal emulated EEPROM.

Simsys, 2025-05-28:
Question about this: How can the frontend recognize beyond doubt that AHRS is blocked? This is necessary in order to write a corresponding message on the horizon display.

A simple solution would be to code it in the sensor box “System state and Git Tag” filegram 0x12a:

For example, bit field
0x0000_0004 AHRS Blocked

realtimepeople, 2025-05-28:
Agree.
This is the wrong place within the CAN address-room anyway.
Looking for a better solution...

realtimepeople, 2025-05-28:
The roll- and pitch CAN packet (ID=0x120) is sent only if the horizon is available.
Is this information eventually sufficient for the frontend ?

Simsys, 2025-05-28:
Other solution please. The absence of a telegram is difficult to detect and can also have other causes.

Simsys, 2025-05-28:
agreed