larus-breeze/sw_sensor #113: NMEA communication with XCSoar

dragonfly771, 2025-03-14:
In combination with an OpenVario or Android navigation device and the Front End Vario Display, a solution should be implemented to enable bidirectional switching of Vario-Speed-to-Fly (STF) mode via the NMEA protocol. Additionally, the protocol could include an option for volume control.

This would allow, for example, adjustments to volume and the MacCready (MC) setting through a menu implemented in XCSoar (using XCI, similar to OpenVario). 

For example, the Vario-Speed-to-Fly (STF) mode switching was implemented in the XCSoar XCVario driver.

realtimepeople, 2025-03-14:
We will implement this together with the current CAN-communication changes.

dragonfly771, 2025-07-01:
One reminder for this issue. It would be great if it works the same way as on the vario display.

MaxBaex, 2025-07-05:
Speed to Fly shall now be forwared from a Larus Frontend via CAN to the Larus Sensor and then via Bluetooth /RS232 to a to the Sensor connected  XCSOAR device.    Implemented with https://github.com/larus-breeze/sw_sensor/issues/169