larus-breeze/sw_sensor #117: Flash Sector and Page Bug in Reprom emulation

MaxBaex, 2025-04-19:
The linker file specifies the following in the flash for the eeprom emulation:
  EEPROM (xrw)	: ORIGIN = 0x080F8000, 	LENGTH = 32K
eeprom.h specifies:
#define EEPROM_START_ADDRESS  ((uint32_t)0x080F8000) /* EEPROM emulation start address:
                                                  from sector2 : after 16KByte of used 
                                                  Flash memory */

These are the last 32KByte of the 1MByte STM32F407 flash. There is a 128kByte and no 16Kbyte Sectors!  Sector 11 0x080E 0000 - 0x080F FFFF 128 Kbytes  (From M0090 Reference manual)   

I believe there must be a bug in the Flash emulation which might cause a problem. As the two 16Kbyte sectors can not be erased individually. 

The uSD_handler uses Flash Sector 8,9 and 10 (384kByte) for intermediate Storage of a new software image in the address range: 0x8080000 - 0x80DFFFF

MaxBaex, 2025-04-22:
The current programs flash utilization is just below 256kB. We should at lease plan 384kB in size to have sufficient reserves.

The Firmware update process as of now supports up to 384kByte in size and temporarily stores the program from the sd-card to flash sector 8,9 and 10 with each 128kB size. Sector 0 ...7 (512k) is reserved for the running program. With this update process sector 7 (128k) will never be used. 

Option 1:
- Restrict program flash location and size to 384k in sector 0 ... 6.
- Use Sector 7,8,9 (384k) for the temporary storage of the flash image.  
- Use Sector 10 and 11 (last 256k of the flash) for the EEPROM emulation.   

Option 2:
- Place program image in sector 3.  
- Place some jump code in sector 0 which jumps to sector 3. 
- Use the smaller sectors 1 and 2 with each 16kB for the EEPROM emulation

I prefer Option 1, as it is simpler and more easy to implement at the cost of more flash memory used for the EEPROM emulation. 

Any objections or other ideas?

realtimepeople, 2025-04-23:
Sure, option 1 fits our needs best.

Simsys, 2025-04-23:
We should check again whether the changes to the Flash layout have an effect on the update mimic.

realtimepeople, 2025-04-23:
If we would really change the FLASH layout we need to make sure that the EEPROM information in already installed sensors is automatically ported to the new location in FLASH.

MaxBaex, 2025-04-23:
There is known bug in the EEPROM emulation Libary, which needs to be fixed: https://github.com/STMicroelectronics/STM32CubeF1/issues/11 

We have about 25 EEPROM Parameters: https://github.com/larus-breeze/sw_algorithms_lib/blob/main/NAV_Algorithms/persistent_data.h  
Most parameters are "32bit float" values which are converted from and to a 16 Bit integer value. This conversion is error prone. They are rarely written during a configuration or after a flight.

Another approach could be to only use the last 128kB flash sector for storing parameters and to use a 32bit size. Writing 128k would take quite some time (~some thousand flights). Once the page is fully written, all 25 parameters are stored in RAM, the sector is erased and the last values are written again. This process would take about  1 ... 4 seconds. This duration is critical, as a program error or a power loss would cause that all configuration parameters are restored to its default value.

MaxBaex, 2025-05-09:
Here the flash sectors:
[image]

MaxBaex, 2025-05-12:
Testcode in NAV_Algorithms/persistent_data.cpp
uint64_t getTime_usec(void);

bool EEPROM_initialize( void)
{
  unsigned status;

  status = HAL_FLASH_Unlock();
  ASSERT(status == HAL_OK);

  status = EE_Init();
  ASSERT(status == HAL_OK);

  //Test the page swapping.  One Page == 128kB.  50% netto storage. One Page requires almost 32768 write to fill completely.
  int i = 0;  //Stop with debugger and modify for testing.
  int numberOfWrites = 0;
  uint64_t max_duration = 0;
  uint64_t min_duration = 0xFFFFFFFFFFFFFFFF;
  uint64_t start = 0;
  uint64_t duration = 0;

  uint16_t value = 0;

  while(i > 0){
      start = getTime_usec();
      EE_WriteVariable(ANT_SLAVE_RIGHT, value);
      duration = getTime_usec() - start;
      if (duration > max_duration){
	  max_duration = duration;
      }

      if (duration < min_duration){
	  min_duration = duration;
      }

      numberOfWrites++;
      value++;
      i--;
  }
  return HAL_OK;
}

MaxBaex, 2025-05-12:
Working EEPROM emulation with page swap is here: https://github.com/larus-breeze/sw_sensor/compare/master...issue_117

Integration in System and Test with Software Update process is not done yet.

MaxBaex, 2025-05-14:
Merge branch issue_75

MaxBaex, 2025-05-15:
**Tested the EEPROM Lib as it is currently.** 
There are sporadic faults retured by EE_WriteVariable and EE_ReadVariable()   These errors immediately disappear if access to these functions is only done from one task context. A simple mutex take & give in the write an read function causes a deadlock during testing.  
**Result**: The lib is not thread save but the applicated required this. Serialization of the API execution needs to be enforced. Better  to encapsulate read and writes from/to flash in a separate task. 

Introduced a ram new layer to serialized writing and reading from flash to a separate task. Public interface only read and writes directly to ram. Public interface simplified and reduced. 
- Removed all EE_Init and Flash Lock calls from the application.  This is now handled in a EEPROM Background task.
- Read operations are always delivered from RAM.
- Write operations are always written to RAM first. The trigger a "written" flag. 
- A background task checks this flag and syncs the value to the flash. This function compares all values in the flash and only writes if they really changed. 
- Read and Write function block until the EEPROM is completely read after system start.  
- The Lib also changed and is now simplified: https://github.com/larus-breeze/sw_algorithms_lib/compare/sw_sensor_issue_117

MaxBaex, 2025-05-15:
#75 will be implemented together with this issue. As the eeprom fix required a Flashlayout change.

MaxBaex, 2025-05-15:
It should be possible to implement an EEPROM migration in EE_task_runnable(). This needs to be done  before calling EE_Init() the first time  as EE_Init will ERASE the entire EEPROM flash area for the new layout. 

How can be robustely ensured that this migration function only runs once?

MaxBaex, 2025-05-17:
Migration implemented and tested when upgrading from 0.4.0

MaxBaex, 2025-05-17:
Waiting for lib merge  https://github.com/larus-breeze/sw_algorithms_lib/pull/54/

MaxBaex, 2025-05-20:
lib merged and issue close with pull  #137