larus-breeze/sw_sensor #129: $PLARV shows value 0 when having no GPS Fix  regardless the pressure speed

Ventusfahrer, 2025-05-12:
Testing the Flash Control of the Vario in the workshop:

Test environment: No GPS reception, the speed was set to 170 km by total pressure.

$PLARV records show no indicate airspeed

Ventusfahrer, 2025-05-19:
Need to be cross-checked. 

Maybe the CAN cable was not working properly. 

The Vario did send 0 - values in this thime.

realtimepeople, 2025-05-21:
The sensor internally monitors the state "airborne"/"on ground".
This information can be published using the existing CAN packet "CAN_Id_Voltage_Circle".
Summarized this means that the sensor can provide a status like "On Ground","Straight flight","Transition","Circling".
Do we want to implement this functionality ?

Ventusfahrer, 2025-06-01:
can be closed, CAT cable was propably not working when performing the flash control test.

closed 2025-06-02 by Ventusfahrer