larus-breeze/sw_sensor #130: Vario value decrease during dynamic thermal entry

DanD222, 2025-05-13:
Hello,

I have a question about the behaviour of the vario during a pull-up in a dynamic thermal entry. 
I've built a Larus for a friend, who describes the following behaviour:

- When cruising at high speed, the vario indicates for example a 2 m/s thermal
- During a pull-up in the thermal, the vario reading drops, to perhaps 1 m/s
- At this point, the pilot needs to decide whether to bank in the thermal or straighten out and continue on course, depending on whether the thermal strength holds. My friend reports that during these moments, he relies on the readings/sound from his LX160 to make the decision.
- Once circling in the thermal, the vario reading shows a reading of 2 m/s again

Has anyone experienced a similar issue / is it possible that this has been addressed since the last release? 

The firmware used is 0.4.0 from August 2024
The IMU sensor is an MTi-1-5A-T

realtimepeople, 2025-05-14:
Dan is absolutely right, there is a peculiarity of the vario-reading when a lifting air mass is entered.
This is due to the energy gain of the vertical speed that is also taken into account by the TEC-mechanism.
It is not possible to remove this effect without impairment of the TEC-precision.
Ideally the peak is exactly twice as high as the strength of the thermal entered.

See https://github.com/larus-breeze/doc_larus/blob/master/helper/vario_lift.png

realtimepeople, 2025-05-14:
I forgot tho mention another effect in my previous post:
If you pull up pretty hard, with as example 2 g-units, the glider transiently experiences a pretty high drag.
This creates a impulse of negative variometer reading fitting to the excess g-load.
Presently research is in progress to describe this effect precisely using an earodynamic model for the glider and the surrounding air mass.

DanD222, 2025-05-14:
Thanks for looking into it; we're not quite sure whether the graph shows a situation of flying at constant airspeed in level flight, and merely experiencing a rising air mass, or whether there are airspeed changes involved too?

The described scenario has the element of the airspeed decreasing during the pull-up; G forces could also play a role.

In either case, the vario reading takes a good few seconds to stabilise after the initial drop in value.

Also, the actual vario reading during the pull-up should probably be the higher value, i.e. 2 m/s, instead of the momentarily-lower 1 m/s, which seems opposite to what I take the graph to mean?

realtimepeople, 2025-05-14:
The graph https://github.com/larus-breeze/doc_larus/blob/master/helper/vario_lift.png is the result of a software-in-the-loop-simulation using synthetic data. Input is a constant horizontal speed all the time and a (unususally hight) lift of 10m/s that is started and ended by a 1g acceleration.

The TEC-variometer inside the algorithms has no delay at all; it's response is only limited by the 100Hz sampling rate.
To provide a smooth reading we usually apply a 1 second delay to the output. This is also the case for the SIL example above.

In the opposite case of a lift triggered only by a pull-up without gain or loss of energy the algorithm delivers a output of zero all the time: https://github.com/larus-breeze/doc_larus/blob/master/helper/vario_pullup.png
Only if we use the 3d (not 2d !) TEC-algorithm, compare the purple and green lines.

In reality the situation is much more complex due to aerodynamic effects and air turbulence.

DanD222, 2025-05-14:
Thanks. So the graphs show the effects of two phenomena (encountering rising air, effect of G force) in isolation; I'm wondering what the graph would look like in the complex interaction of say the following?

- Initial speed of 160 km/h
- Encountering rising air of 2 m/s
- 2 second delay
- 2G pull up
- Decrease of speed to 80 km/h

realtimepeople, 2025-05-15:
When I find the time I will create a simulation for this scenario.
But the outcome is clear:
The beginning looks like the lift scenario above.
For the rest of the time the vario will stay stable at 2m/s.
But this is true for a friction-free case only.
In reality you will see a vario dip due to the increased drag while pulling up.

realtimepeople, 2025-05-15:
About the vario-dip when pulling up:
I have made a test-flight to find out the effect of g-load on the variometer-reading.
The lesson learned is that we probably need a acceleration-compensation additionally to the speed-compensation.
Look here: https://github.com/larus-breeze/doc_larus/blob/master/helper/G-load_wobble.png
The drag is visible as a negative vario-reading.
It is increased momentarily when a high OR a low g-load is applied !