larus-breeze/sw_sensor #132: Incorrect wind speed after the first thermaling

Ventusfahrer, 2025-05-17:
[image]

The problem has been observed a total of 3 times (2 x 7v, 1 x DU) with slightly different characteristics:
1. wind speed unrealistic but still within the expected range (<120 km/h) (observed once)
2. wind speed (see photo) completely unrealistic (observed twice)

The occurrence of the situation is very likely limited to the first circling after take-off.

If 1. occurs, the glider will heal itself within a few minutes. In the case of characteristic 2, the wind speeds fall slowly, but far too slowly for self-healing due to the very large difference to the actual measured values.

Switching the sensor box and vario on/off immediately solved the problem (2nd characteristic). The vario unit then ran faultlessly for the next 7:40 hours.

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

Ventusfahrer, 2025-05-17:
next test, new calibration, no chance to get the Vario System working, even several restarts did not show any effect.

EEPROM-Files: https://sfg.daten-cloud.de/s/EMyxafLnA8RiAym

realtimepeople, 2025-05-21:
This is a communication issue of the system, not a problem of the sensor firmware.
The wind measurement is most precise when circling as almost all measurement errors cancel out when averaging over a complete circle.

closed 2025-05-21 by realtimepeople