larus-breeze/sw_sensor #139: 0.5.0 *pre-release* does not run

Ventusfahrer, 2025-05-23:
Before Klaus forgets to report it:

When the .elf file is flashed, the sensor does not start.

Workaround: install 0.4.0 .elf file and update to 0.5.0 version via uSD card.

MaxBaex, 2025-05-23:
Retested with the Cube Programmer via the USB-DFU Bootloader mode. The sensor starts in every case a power cycle after flashing in order to leave the DFU Bootloade Mode. Tested the following.

- Without an SD-Card and some valid EEPROM Data from a previously flashed 0.5.0.  Flashed 0.5.0 elf image with 
   -  only erased sectors 0 ... 5
   -  complete mass erase before flashing

- Retested with inserted SD-Card without a larus_sensor_config.ini file on the sd card
   -  only erased sectors 0 ... 5
   -  complete mass erase before flashing

- Retested with a larus_sensor_config.ini file on the inserted sd card
   -  only erased sectors 0 ... 5
   -  complete mass erase before flashing 

Can not reproduce the issue / bug.

MaxBaex, 2025-06-03:
Closing as this can not be reproduced.