larus-breeze/sw_sensor #144: Partially implausible values of the “Essential” version

Simsys, 2025-05-30:
Some pilots of SG Bensheim and also some SteFly customers have reported problems with Larus Version Essential.

Wind speeds of more than 1000 km/h are being reported and the vario display is spinning and jumping back and forth wildly.

What we know:
a) The problem only occurs with the Essential version
b) The problem is only visible in flight, if you look at the data in the PC simulation, you will not see the error.
c) The problem occurs on some flights, but not on other flights on the same hardware and software with the same aircraft.
d) Restarting the system during the flight sometimes solves the problem

Example:

[image]

MaxBaex, 2025-06-03:
The clipping to 0.85 causes that flying above ~3500m of altitude uses a wrong density. There shall be a plan and issue to remove this  workaround again if merged and released. I think it would be better to fall back to the ICAO Standard Atmosphere Density if the calculated density is to far off.

MaxBaex, 2025-06-04:
Please retest this problem with [the](https://github.com/larus-breeze/sw_sensor/releases/tag/0.5.1)   and confirm that it is resolved.

dragonfly771, 2025-06-08:
What would happen if a pressure sensor is defective? We once had a Larus sensor with a broken MS5611 sensor and it took some time to figure out that this was the problem.

Simsys, 2025-10-22:
Solved with [0.5.3](https://github.com/larus-breeze/sw_sensor/releases/tag/0.5.3)

closed 2025-10-22 by Simsys