larus-breeze/sw_sensor #159: Add extra magnetometer to improve the single-GNSS variant

realtimepeople, 2025-06-14:
Meanwhile we have quite a lot of experience with Larus installations in many gliders.

It becomes more and more clear that mounting the sensor close to or inside of the instrument panel
is favorable because we only need short wires and tubes making the installation simple and fast.

Unfortunately, this place is a bad one for the magnetometer.
In many cases there are power lines with varying currents close to the induction sensor.
This does drastically impact the performance of the magnetic compass.

Example: 
A wire carrying 1A at a distance of 10cm to the induction sensor creates a distortion
of the earth induction corresponding to 3 degrees error in magnetic heading.
So the feed lines of canopy flashers or of the radio greatly downgrades the precision of the AHRS.

We shall add an extra hardware component: 
An extra magnetometer that can be mounted in a place with no power-lines in the vicinity.
This sensor shall be connected to the Larus system using the CAN bus.

We do already have the algorithms to calibrate the orientation and offset/gain of such an extra sensor
automatically. So the implementation of the extra algorithms shall be quite simple.

Simsys, 2025-06-14:
In my opinion, the approach should be to really understand the problem. To do this, you have to bring the problem into the lab, where you can reproduce it.

The first step is to collect all relevant data; a proposal for this is already on the table (see #157).

It is quite likely that the problem can be solved by better algorithms or by fixing bugs.

dragonfly771, 2025-07-01:
In general it sounds like a good idea but the precise installation of the (small) magnetic sensor will be also difficult and may induce some more errors. There are indeed some difficulties with the installation of the Essential Larus in some gliders and the satisfaction level of D-GNSS users is higher. I´m not sure if the external magnetic sensor solves a lot of problems with bad installations. On the other hand, if it is possible without too much work and the hardware costs are manageable, we should just try it.

Simsys, 2025-07-01:
In the end, it takes a lot of work to get such a product ready for series production. The device has to be designed and the interfaces defined, hardware, housing and software have to be developed. Finally, documentation for marketing and installation, including error situations, must be available.

I also don't see the benefit. The sensor box is already a small remote sensor that can be placed anywhere. How should a second sensor box help us? It will only make everything more complicated. The quantities will be negligible.

We have so many unfinished work to do. We urgently need to improve the reliability of the system. We keep seeing glitches in the functionality, especially in the essential version, but also in the dual GNSS version. In the event of a GNSS failure, we need a good fallback to a decent standard vario.

We shouldn't waste our limited developer resources. I advise against starting this project.